mathieugouin/rpiDvr #21: Entering folder with large number of photos crash kodi

mathieugouin, 2017-01-16:
Go to:
* Top menu
* Pictures
* USB Mounted Ext HD
* Navigate to the large folder

At this point, the list of files are briefly displayed, then kodi crash.  It restarts automatically.

mathieugouin, 2017-01-19:
After quick investigation, there seem to have corrupted jpeg in the picture library.

mathieugouin, 2017-01-19:
The crash can also be reproduced when opening the problematic JPEG from the file manager.

mathieugouin, 2017-01-20:
Check if related to progressive jpg not supported in GPU decoding.

mathieugouin, 2017-01-20:
https://www.raspberrypi.org/forums/viewtopic.php?f=70&t=92401

mathieugouin, 2017-01-21:
Currently, affected images are ```DSC*.jpg```

Sony camera.

mathieugouin, 2017-01-21:
Sony DSC-W510

mathieugouin, 2017-01-21:
Camera make : SONY

Camera model : DSC-W510

Date/Time :2012/05/13 16:51:13

Resolution : 4000 x 3000

Flash used : No

Focal length : 5.6mm (35mm equivalent 31mm)

Exposure time : 0.0020 s (1/500)

Aperture : f/3.0

ISO equiv. : 80

Whitebalance : Auto

Metering Mode : multi spot

Exposure : program (auto)

mathieugouin, 2017-01-22:
[kodi.old.log.txt](https://github.com/mathieugouin/rpiDvr/files/721619/kodi.old.log.txt)

mathieugouin, 2017-01-22:
https://paste.pound-python.org/show/CgaPIsie0lMepq2XmJjP/

mathieugouin, 2017-01-26:
Refer to : http://forum.kodi.tv/showthread.php?tid=287441

mathieugouin, 2017-04-27:
Need to try exiftool to remove metadata?

mathieugouin, 2017-04-27:
Also http://forum.kodi.tv/showthread.php?tid=251908

mathieugouin, 2017-04-28:
Check http://exif.regex.info/exif.cgi

mathieugouin, 2017-04-29:
Working:
[image]

mathieugouin, 2017-04-29:
Bad camera 1: 
[image]

mathieugouin, 2017-04-29:
Bad camera 2: 
[image]

mathieugouin, 2017-04-29:
Need to re-submit debug log with repository cleaned-up (clean-up done)

mathieugouin, 2017-04-29:
* https://picload.org/image/rciggpww/p2180004.jpg
* https://picload.org/image/rciggwrr/p2180006.jpg
* https://picload.org/image/rciggwda/p2180003.jpg

mathieugouin, 2017-04-29:
Camera: Sony DSC-W510
* https://picload.org/image/rolpocll/dsc00166.jpg
* https://picload.org/image/rolpocgr/dsc00167.jpg
* https://picload.org/image/rolpocra/dsc00168.jpg
* https://picload.org/image/rolpocdi/dsc00196.jpg
* https://picload.org/image/rolpocor/dsc00198.jpg

sadortun, 2017-04-29:
Je sais pas de quand sont ses photos, mais tout mes félicitations à toi et ta femme ! !

mathieugouin, 2017-04-29:
Salut hihihi...

Merci! J'espère que tout va bien de ton côté!

C'est drôle mais la photo de bedaine est pour mon premier et la photo de
bébé c'est mon deuxième :)

Je ne me rappelais plus que tu étais notifié sur mes notes... Hahah...

Peut être que tu pourrais m'aider? Tu connais un analyseur de JPEG?  J'ai
trouvé http://exif.regex.info/exif.cgi

Je veux comprendre ce qu'il y a de différent entre 2 photos. Pas l'image
mais plus le meta data.

À bientôt!
Mathieu

mathieugouin, 2017-04-29:
Another debug log without banned repository:

http://paste.ubuntu.com/24478887/

mathieugouin, 2017-11-04:
Found also an Olympus camera that create the same problem with all JPG.  More data to come...

mathieugouin, 2017-11-04:
Found another Sony camera with the same issue.

mathieugouin, 2020-01-23:
Solved w/ Kodi 18.5 using libreelec 9.2.0

closed 2020-01-23 by mathieugouin